a9183756-gh/Arduino-CMake-Toolchain #7: ESP8266

Jonas18175, 2020-03-24:
Hello, nice work to bring a actual version of a cmake-toolchain variant.

I like to use this with arduino + esp8266 so i tried to setup a project with this, but i can't link it.

Look at following terminal output:
```
Executing recipe.hooks.linking.prelink.1.pattern hook
/home/jonas/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: warning: cannot find entry symbol app_entry; defaulting to 0000000040100000
/home/jonas/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /home/jonas/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/sdk/libc/xtensa-lx106-elf/lib/libc.a(lib_a-dtoa.o):(.literal+0x44): undefined reference to `malloc'
/home/jonas/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /home/jonas/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/sdk/libc/xtensa-lx106-elf/lib/libc.a(lib_a-dtoa.o): in function `_dtoa_r':
/home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdlib/dtoa.c:238: undefined reference to `malloc'
/home/jonas/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /home/jonas/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/sdk/libc/xtensa-lx106-elf/lib/libc.a(lib_a-mprec.o):(.literal+0x0): undefined reference to `_calloc_r'
/home/jonas/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /home/jonas/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/sdk/libc/xtensa-lx106-elf/lib/libc.a(lib_a-mprec.o): in function `_Balloc':
/home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdlib/mprec.c:103: undefined reference to `malloc'
/home/jonas/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdlib/mprec.c:107: undefined reference to `_calloc_r'
/home/jonas/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdlib/mprec.c:124: undefined reference to `_calloc_r'
/home/jonas/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /home/jonas/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/sdk/libc/xtensa-lx106-elf/lib/libc.a(lib_a-mprec.o): in function `_Bfree':
/home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdlib/mprec.c:139: undefined reference to `malloc'
/home/jonas/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /home/jonas/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/sdk/libc/xtensa-lx106-elf/lib/libc.a(lib_a-mprec.o): in function `__pow5mult':
/home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdlib/mprec.c:435: undefined reference to `malloc'
/home/jonas/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /home/jonas/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/sdk/libc/xtensa-lx106-elf/lib/libc.a(lib_a-nano-svfprintf.o):(.literal+0x0): undefined reference to `_malloc_r'
/home/jonas/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /home/jonas/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/sdk/libc/xtensa-lx106-elf/lib/libc.a(lib_a-nano-svfprintf.o):(.literal+0x4): undefined reference to `_realloc_r'
/home/jonas/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /home/jonas/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/sdk/libc/xtensa-lx106-elf/lib/libc.a(lib_a-nano-svfprintf.o):(.literal+0x8): undefined reference to `_free_r'
/home/jonas/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /home/jonas/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/sdk/libc/xtensa-lx106-elf/lib/libc.a(lib_a-nano-svfprintf.o): in function `__ssputs_r':
/home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c:202: undefined reference to `_malloc_r'
/home/jonas/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c:213: undefined reference to `_realloc_r'
/home/jonas/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c:217: undefined reference to `_free_r'
/home/jonas/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /home/jonas/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/sdk/libc/xtensa-lx106-elf/lib/libc.a(lib_a-nano-svfprintf.o): in function `__ssprint_r':
/home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c:293: undefined reference to `_malloc_r'
/home/jonas/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c:304: undefined reference to `_realloc_r'
/home/jonas/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c:309: undefined reference to `_free_r'
/home/jonas/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /home/jonas/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/sdk/libc/xtensa-lx106-elf/lib/libc.a(lib_a-nano-svfprintf.o): in function `_svfprintf_r':
/home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c:507: undefined reference to `_malloc_r'
collect2: error: ld returned 1 exit status
make[3]: *** [CMakeFiles/IOTNetCore.dir/build.make:92: IOTNetCore.elf] Fehler 1
make[3]: Verzeichnis „/home/jonas/DEV/Projects/DEV.CPP.IOTNetCore/cmake-build-debug“ wird verlassen
make[2]: *** [CMakeFiles/Makefile2:170: CMakeFiles/IOTNetCore.dir/all] Fehler 2
make[2]: Verzeichnis „/home/jonas/DEV/Projects/DEV.CPP.IOTNetCore/cmake-build-debug“ wird verlassen
make[1]: *** [CMakeFiles/Makefile2:177: CMakeFiles/IOTNetCore.dir/rule] Fehler 2
make[1]: Verzeichnis „/home/jonas/DEV/Projects/DEV.CPP.IOTNetCore/cmake-build-debug“ wird verlassen
make: *** [Makefile:157: IOTNetCore] Fehler 2
```

a9183756-gh, 2020-03-25:
ESP8266 (Version 2.6.3) works for me very well, and I am surprised to see these linker errors for you.
Can you please help with the following
1. Post logs with verbose build (make VERBOSE=1).
2. Try to build the "Examples/01_hello_world" source and see if you see the same errors.
3. As given in some google references (https://github.com/esp8266/Arduino/issues/11), ESP8266 libc does not seem to have malloc/free etc. and has similar functions like os_malloc/os_free etc. You may try replacing with ESP8266 specific equivalent functions in your source. If portability is required, you can use the preprocessor flag ARDUINO_ARCH_ESP8266 for such ESP8266 specific code.

a9183756-gh, 2020-03-25:
Also please help with the following
    4. From Arduino IDE, compile 'Blink' example and post the logs. Before the build, enable verbose output (From the menu File->Preferences->Show verbose output during->compilation).
The above step will help to compare the linking command with that of Arduino IDE, if the linking works in Arduino IDE.

Jonas18175, 2020-03-25:
I tried to compile it with arduino and it is the same error output:
```
Arduino: 1.8.12 (Linux), Board: "Generic ESP8266 Module, 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), dtr (aka nodemcu), 26 MHz, 40MHz, DOUT (compatible), 1MB (FS:64KB OTA:~470KB), 2, nonos-sdk 2.2.1+100 (190703), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

/usr/share/arduino/arduino-builder -dump-prefs -logger=machine -hardware /usr/share/arduino/hardware -hardware /home/jonas/.arduino15/packages -tools /usr/share/arduino/tools-builder -tools /home/jonas/.arduino15/packages -libraries /home/jonas/Arduino/libraries -fqbn=esp8266:esp8266:generic:xtal=80,vt=flash,exception=legacy,ssl=all,ResetMethod=nodemcu,CrystalFreq=26,FlashFreq=40,FlashMode=dout,eesz=1M64,led=2,sdk=nonosdk_190703,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=115200 -ide-version=10812 -build-path /tmp/arduino_build_868128 -warnings=none -build-cache /tmp/arduino_cache_380249 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mklittlefs.path=/home/jonas/.arduino15/packages/esp8266/tools/mklittlefs/2.5.0-4-69bd9e6 -prefs=runtime.tools.mklittlefs-2.5.0-4-69bd9e6.path=/home/jonas/.arduino15/packages/esp8266/tools/mklittlefs/2.5.0-4-69bd9e6 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=/home/jonas/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-4-b40a506.path=/home/jonas/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506 -prefs=runtime.tools.mkspiffs.path=/home/jonas/.arduino15/packages/esp8266/tools/mkspiffs/2.5.0-4-b40a506 -prefs=runtime.tools.mkspiffs-2.5.0-4-b40a506.path=/home/jonas/.arduino15/packages/esp8266/tools/mkspiffs/2.5.0-4-b40a506 -prefs=runtime.tools.python3.path=/home/jonas/.arduino15/packages/esp8266/tools/python3/3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=/home/jonas/.arduino15/packages/esp8266/tools/python3/3.7.2-post1 -verbose /tmp/arduino_modified_sketch_271734/sketch_mar25a.ino
/usr/share/arduino/arduino-builder -compile -logger=machine -hardware /usr/share/arduino/hardware -hardware /home/jonas/.arduino15/packages -tools /usr/share/arduino/tools-builder -tools /home/jonas/.arduino15/packages -libraries /home/jonas/Arduino/libraries -fqbn=esp8266:esp8266:generic:xtal=80,vt=flash,exception=legacy,ssl=all,ResetMethod=nodemcu,CrystalFreq=26,FlashFreq=40,FlashMode=dout,eesz=1M64,led=2,sdk=nonosdk_190703,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=115200 -ide-version=10812 -build-path /tmp/arduino_build_868128 -warnings=none -build-cache /tmp/arduino_cache_380249 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mklittlefs.path=/home/jonas/.arduino15/packages/esp8266/tools/mklittlefs/2.5.0-4-69bd9e6 -prefs=runtime.tools.mklittlefs-2.5.0-4-69bd9e6.path=/home/jonas/.arduino15/packages/esp8266/tools/mklittlefs/2.5.0-4-69bd9e6 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=/home/jonas/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-4-b40a506.path=/home/jonas/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506 -prefs=runtime.tools.mkspiffs.path=/home/jonas/.arduino15/packages/esp8266/tools/mkspiffs/2.5.0-4-b40a506 -prefs=runtime.tools.mkspiffs-2.5.0-4-b40a506.path=/home/jonas/.arduino15/packages/esp8266/tools/mkspiffs/2.5.0-4-b40a506 -prefs=runtime.tools.python3.path=/home/jonas/.arduino15/packages/esp8266/tools/python3/3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=/home/jonas/.arduino15/packages/esp8266/tools/python3/3.7.2-post1 -verbose /tmp/arduino_modified_sketch_271734/sketch_mar25a.ino
Using board 'generic' from platform in folder: /home/jonas/.arduino15/packages/esp8266/hardware/esp8266/2.6.3
Using core 'esp8266' from platform in folder: /home/jonas/.arduino15/packages/esp8266/hardware/esp8266/2.6.3
Detecting libraries used...
/home/jonas/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -I/home/jonas/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/sdk/include -I/home/jonas/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/sdk/lwip2/include -I/home/jonas/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/sdk/libc/xtensa-lx106-elf/include -I/tmp/arduino_build_868128/core -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++11 -ffunction-sections -fdata-sections -fno-exceptions -w -x c++ -E -CC -DNONOSDK22x_190703=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10812 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_GENERIC\"" -DLED_BUILTIN=2 -DFLASHMODE_DOUT -DESP8266 -I/home/jonas/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266 -I/home/jonas/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/variants/generic /tmp/arduino_build_868128/sketch/sketch_mar25a.ino.cpp -o /dev/null
Generating function prototypes...
/home/jonas/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -I/home/jonas/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/sdk/include -I/home/jonas/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/sdk/lwip2/include -I/home/jonas/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/sdk/libc/xtensa-lx106-elf/include -I/tmp/ardui [... cut ...] tools/sdk/libc/xtensa-lx106-elf/lib/libc.a(lib_a-nano-svfprintf.o): in function `_svfprintf_r':
/home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c:507: undefined reference to `_malloc_r'
collect2: error: ld returned 1 exit status
exit status 1
Fehler beim Kompilieren für das Board Generic ESP8266 Module.
```
The error is not from your cmake project, but i had alredy compiled source for the esp8266 and i flashed it with the Arduino IDE. I have no idea what is wrong.

a9183756-gh, 2020-03-26:
The linker scripts for the ESP8266 might be behaving incorrectly. Try re-installing ESP8266. If you still face the issue, please  get the help of ESP8266 experts. I will close this issue for now.

closed 2020-03-26 by a9183756-gh