a9183756-gh/Arduino-CMake-Toolchain #8: Cannot find Arduino Nano

blasco, 2020-03-25:
It seems to be looking at the right places:

```
-- Found Arduino Platform: /home/blasco/.arduino15/packages/esp32/hardware/esp32/1.0.2
-- Found Arduino Platform: /home/blasco/.arduino15/packages/esp8266/hardware/esp8266/2.5.0
-- Found Arduino Platform: /home/blasco/.arduino15/packages/arduino/hardware/avr/1.6.23
-- Found Arduino Platform: /home/blasco/.arduino15/packages/arduino/hardware/megaavr/1.8.3
-- Found Arduino Platform: /home/blasco/.arduino15/packages/atmel-avr-xminis/hardware/avr/0.
```

I can see that there is a Arduino Nano option in boards.txt for 
```
-- Found Arduino Platform: /home/blasco/.arduino15/packages/arduino/hardware/avr/1.6.23
```
And I can see the arduino nano in board.txt there:
```
nano.name=Arduino Nano

nano.upload.tool=avrdude
nano.upload.protocol=arduino

nano.bootloader.tool=avrdude
nano.bootloader.unlock_bits=0x3F
nano.bootloader.lock_bits=0x0F

nano.build.f_cpu=16000000L
nano.build.board=AVR_NANO
nano.build.core=arduino
nano.build.variant=eightanaloginputs
```
Nevertheless, I don't see arduino nano related content in platform.txt

Is the script currently looking only at platform txt? What do I need to change to make it work with nano? Thank you for the great project

blasco, 2020-03-25:
This is the generated file, which doesn't contain any Arduino Nano entry:
```
# Copyright (c) 2020 Arduino CMake Toolchain

###############################################################################
# This is an automatically generated template file for board options.
# You may edit it to comment/uncomment selected board and board options.
# However do not change the structure of this template, which is fixed and 
# any change to the structure gets overwritten.

#!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
# Arduino boards.
# Platform: esp32
#!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
# set(ARDUINO_BOARD "ESP32 Dev Module [esp32.esp32]") # ESP32 Dev Module
# set(ARDUINO_BOARD "ESP32 Wrover Module [esp32.esp32wrover]") # ESP32 Wrover Module
# set(ARDUINO_BOARD "ESP32 Pico Kit [esp32.pico32]") # ESP32 Pico Kit
# set(ARDUINO_BOARD "Turta IoT Node [esp32.turta_iot_node]") # Turta IoT Node
# set(ARDUINO_BOARD "TTGO LoRa32-OLED V1 [esp32.ttgo-lora32-v1]") # TTGO LoRa32-OLED V1
# set(ARDUINO_BOARD "XinaBox CW02 [esp32.cw02]") # XinaBox CW02
# set(ARDUINO_BOARD "SparkFun ESP32 Thing [esp32.esp32thing]") # SparkFun ESP32 Thing
# set(ARDUINO_BOARD "u-blox NINA-W10 series (ESP32) [esp32.nina_w10]") # u-blox NINA-W10 series (ESP32)
# set(ARDUINO_BOARD "Widora AIR [esp32.widora-air]") # Widora AIR
# set(ARDUINO_BOARD "Electronic SweetPeas - ESP320 [esp32.esp320]") # Electronic SweetPeas - ESP320
# set(ARDUINO_BOARD "Nano32 [esp32.nano32]") # Nano32
# set(ARDUINO_BOARD "LOLIN D32 [esp32.d32]") # LOLIN D32
# set(ARDUINO_BOARD "LOLIN D32 PRO [esp32.d32_pro]") # LOLIN D32 PRO
# set(ARDUINO_BOARD "WEMOS LOLIN32 [esp32.lolin32]") # WEMOS LOLIN32
# set(ARDUINO_BOARD "Dongsen Tech Pocket 32 [esp32.pocket_32]") # Dongsen Tech Pocket 32
# set(ARDUINO_BOARD ""WeMos" WiFi&Bluetooth Battery [esp32.WeMosBat]") # "WeMos" WiFi&Bluetooth Battery
# set(ARDUINO_BOARD "ESPea32 [esp32.espea32]") # ESPea32
# set(ARDUINO_BOARD "Noduino Quantum [esp32.quantum]") # Noduino Quantum
# set(ARDUINO_BOARD "Node32s [esp32.node32s]") # Node32s
# set(ARDUINO_BOARD "Hornbill ESP32 Dev [esp32.hornbill32dev]") # Hornbill ESP32 Dev
# set(ARDUINO_BOARD "Hornbill ESP32 Minima [esp32.hornbill32minima]") # Hornbill ESP32 Minima
# set(ARDUINO_BOARD "FireBeetle-ESP32 [esp32.firebeetle32]") # FireBeetle-ESP32
# set(ARDUINO_BOARD "IntoRobot Fig [esp32.intorobot-fig]") # IntoRobot Fig
# set(ARDUINO_BOARD "Onehorse ESP32 Dev Module [esp32.onehorse32dev]") # Onehorse ESP32 Dev Module
# set(ARDUINO_BOARD "Adafruit ESP32 Feather [esp32.featheresp32]") # Adafruit ESP32 Feather
# set(ARDUINO_BOARD "NodeMCU-32S [esp32.nodemcu-32s]") # NodeMCU-32S
# set(ARDUINO_BOARD "MH ET LIVE ESP32DevKIT [esp32.mhetesp32devkit]") # MH ET LIVE ESP32DevKIT
# set(ARDUINO_BOARD "MH ET LIVE ESP32MiniKit [esp32.mhetesp32minikit]") # MH ET LIVE ESP32MiniKit
# set(ARDUINO_BOARD "ESP32vn IoT Uno [esp32.esp32vn-iot-uno]") # ESP32vn IoT Uno
# set(ARDUINO_BOARD "DOIT ESP32 DEVKIT V1 [esp32.esp32doit-devkit-v1]") # DOIT ESP32 DEVKIT V1
# set(ARDUINO_BOARD "OLIMEX ESP32-EVB [esp32.esp32-evb]") # OLIMEX ESP32-EVB
# set(ARDUINO_BOARD "OLIMEX ESP32-GATEWAY [esp32.esp32-gateway]") # OLIMEX ESP32-GATEWAY
# set(ARDUINO_BOARD "OLIMEX ESP32-PoE [esp32.esp32-poe]") # OLIMEX ESP32-PoE
# set(ARDUINO_BOARD "ThaiEasyElec's ESPino32 [esp32.espino32]") # ThaiEasyElec's ESPino32
# set(ARDUINO_BOARD "M5Stack-Core-ESP32 [esp32.m5stack-core-esp32]") # M5Stack-Core-ESP32
# set(ARDUINO_BOARD "M5Stack-FIRE [esp32.m5stack-fire]") # M5Stack-FIRE
# set(ARDUINO_BOARD "M5Stick-C [esp32.m5stick-c]") # M5Stick-C
# set(ARDUINO_BOARD "ODROID ESP32 [esp32.odroid_esp32]") # ODROID ESP32
# set(ARDUINO_BOARD "Heltec WiFi Kit 32 [esp32.heltec_wifi_kit_32]") # Heltec WiFi Kit 32
# set(ARDUINO_BOARD "Heltec WiFi LoRa 32 [esp32.heltec_wifi_lora_32]") # Heltec WiFi LoRa 32
# set(ARDUINO_BOARD "Heltec WiFi LoRa 32(V2) [esp32.heltec_wifi_lora_32_V2]") # Heltec WiFi LoRa 32(V2)
# set(ARDUINO_BOARD "Heltec Wireless Stick [esp32.heltec_wireless_stick]") # Heltec Wireless Stick
# set(ARDUINO_BOARD "ESPectro32 [esp32.espectro32]") # ESPectro32
# set(ARDUINO_BOARD "Microduino-CoreESP32 [esp32.CoreESP32]") # Microduino-CoreESP32
# set(ARDUINO_BOARD "ALKS ESP32 [esp32.alksesp32]") # ALKS ESP32
# set(ARDUINO_BOARD "WiPy 3.0 [esp32.wipy3]") # WiPy 3.0
# set(ARDUINO_BOARD "BPI-BIT [esp32.bpi-bit]") # BPI-BIT
# set(ARDUINO_BOARD "Silicognition wESP32 [esp32.wesp32]") # Silicognition wESP32
# set(ARDUINO_BOARD "T-Beam [esp32.t-beam]") # T-Beam
# set(ARDUINO_BOARD "D-duino-32 [esp32.d-duino-32]") # D-duino-32
# set(ARDUINO_BOARD "LoPy [esp32.lopy]") # LoPy
# set(ARDUINO_BOARD "LoPy4 [esp32.lopy4]") # LoPy4
# set(ARDUINO_BOARD "OROCA EduBot [esp32.oroca_edubot]") # OROCA EduBot
# set(ARDUINO_BOARD "ESP32 FM DevKit [esp32.fm-devkit]") # ESP32 FM DevKit
# set(ARDUINO_BOARD "Frog Board ESP32 [esp32.frogboard]") # Frog Board ESP32
# set(ARDUINO_BOARD "AI Thinker ESP32-CAM [esp32.esp32cam]") # AI Thinker ESP32-CAM
[... 37 lines cut ...]
#!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
# Arduino boards.
# Platform: Atmel AVR Xplained-minis
#!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
# set(ARDUINO_BOARD "Atmel atmega328p Xplained mini [avr.atmega328p_xplained_mini]") # Atmel atmega328p Xplained mini
[... cut ...]
# Platform: Atmel AVR Xplained-minis
#!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
# set(ARDUINO_BOARD "Atmel atmega328p Xplained mini [avr.atmega328p_xplained_mini]") # Atmel atmega328p Xplained mini
# set(ARDUINO_BOARD "Atmel atmega328pb Xplained mini [avr.atmega328pb_xplained_mini]") # Atmel atmega328pb Xplained mini
# set(ARDUINO_BOARD "Atmel atmega168pb Xplained mini [avr.atmega168pb_xplained_mini]") # Atmel atmega168pb Xplained mini

Menu options etc...
```

blasco, 2020-03-25:
Now that I paste the file, I see that it seems to find all platforms but the ```avr/1.6.23``` one

a9183756-gh, 2020-03-25:
I have fixed this issue, but still working on a stable commit that can be pushed.

The problem is due to "Atmel AVR Xplained-minis" discovery overwriting the previous "AVR" discovery. That is the reason why BoardOptions.txt contains duplicate sections for "Atmel AVR Xplained-minis".

Until I push the fix for this issue in few days, please rename the folder /home/blasco/.arduino15/packages/atmel-avr-xminis to something like /home/blasco/.arduino15/packages/atmel-avr-xminis.bak.

a9183756-gh, 2020-03-27:
This issue is fixed in the current master branch. Please reopen if you still face issue.

closed 2020-03-27 by a9183756-gh

UltraBlackLinux, 2021-08-11:
@a9183756-gh 
Quick somewhat related question - I just got myself an Arduino Nano with an Atmel M328P - Can I just choose the Atmega328P one?