a9183756-gh/Arduino-CMake-Toolchain #17: Feedback

raphael-bmec-co, 2020-04-23:
Hello.

Firstly, thank you for this amazing tool.

Secondly, I have just worked my way through getting this working and I would like to add a little feedback that will make the instructions easier for someone who is potentially inexperienced to follow:

1. I think it would be worth explaining that the CMakeLists.txt from the respective examples should be moved to the root.
2. I'm using CLion and I needed to add the following to the CMake options (the second -D was critical) 

-D CMAKE_TOOLCHAIN_FILE=Arduino-toolchain.cmake -D SERIAL_PORT=COM9

a9183756-gh, 2020-04-23:
Hey, Thanks for your feedback.

For 2, I understand your problem. You are actually using some "undocumented" feature to set the default value of the serial port when invoking the CMake. You can also use the documented way of passing the serial port as environment variable in CLion. However, CLion or other IDE specific usage documentation is lacking and expected to improve in the future.  I suggest you continue using the method you discovered. Note that in release 1.1 of the toolchain, the name changed to ARDUINO_UPLOAD_SERIAL_PORT.

For 1, I did not understand your problem. As far as I know, there is no need to move CMakeLists.txt anywhere. Just need to point to the appropriate directory as the CMake source directory (i.e. Use `Examples` as CMake source directory to build all examples, or use `Examples/01_hello_world` to build only the hello world example.)

raphael-bmec-co, 2020-04-23:
Hey, thanks for getting back to me so quickly.

I was running off master. I have just moved to release 1.1 and I am not having "issue" 1 anymore. I suspect this has more to do with something I changed on my side than the version change.

For the sake of helping anyone who tries to set this up on CLion in the future. The "tricks" for setting up a working project are as follows:

1. Extract the "Arduino-CMake-Toolchain-release-1.1-dev" to wherever you would like to keep the toolchain.
2. Create a folder for your project and copy the example of your choice (for example `CMakeLists.txt` `hello_world.cpp`).
3. Open CLion, and open the project folder. It will automatically load the CMake project and fail.
4. In the File -> Settings -> Build, Execution, Development -> CMake dialog set the CMake options to `-D CMAKE_TOOLCHAIN_FILE=path/to/Arduino-toolchain.cmake -D ARDUINO_UPLOAD_SERIAL_PORT=COM9` (Check the actual port for your device).
5. Click OK. The build will fail as per the documentation to allow selection of the board.
6. Build again and upload as per the documentation.

The most important point is that you should not open the root "Arduino-CMake-Toolchain-release-1.1-dev" in CLion as it will create additional CMakeLists.txt files that will cause issues.

raphael-bmec-co, 2020-04-24:
Hi. Next challenge:

I am struggling to include BLE libraries. Interestingly this is one library with several includes. Maybe that is related to the issue?

I have tried replacing `BLEDevice ` with the library name `ESP32 BLE Arduino`.
I have tried the other includes: `BLEUtils ` and `BLEServer`.

I have also tried `AUTO_PUBLIC`, `AUTO_PRIVATE`, `PUBLIC `and `PRIVATE`.

CMakeList.txt:
```
#=================================================================
# app
add_executable(arc003-2 arc003-2.cpp)

target_link_arduino_libraries(arc003-2 PRIVATE BLEDevice core)
# This is needed for the generation of HEX binary and uploading it
target_enable_arduino_upload(arc003-2)

#=================================================================
# libs
add_library(serial_debug utils/serial_debug/serial_debug.h)
target_link_arduino_libraries(serial_debug AUTO_INTERFACE utils/serial_debug/serial_debug.h AUTO_PRIVATE)
```

Error:
```
"C:\Program Files\JetBrains\CLion 2020.1\bin\cmake\win\bin\cmake.exe" -DCMAKE_BUILD_TYPE=Debug -D CMAKE_TOOLCHAIN_FILE=C:/workspace/arduino/Arduino-CMake-Toolchain-release-1.1-dev/Arduino-toolchain.cmake -D ARDUINO_UPLOAD_SERIAL_PORT=COM9 -G "CodeBlocks - MinGW Makefiles" C:\workspace\ARC003-2
CMake Error at C:/workspace/arduino/Arduino-CMake-Toolchain-release-1.1-dev/Arduino/System/BoardBuildTargets.cmake:433 (message):
  Arduino library BLEDevice could not be found in
  C:/workspace/ARC003-2;C:/workspace/ARC003-2;C:/workspace/ARC003-2/cmake-build-debug/_pkg_mgr;C:/Users/Raph/Documents/Arduino;C:/Users/Raph/AppData/Local/Arduino15/packages/esp32/hardware/esp32/1.0.4;C:/Program
  Files (x86)/Arduino
Call Stack (most recent call first):
  C:/workspace/arduino/Arduino-CMake-Toolchain-release-1.1-dev/Arduino/System/BoardBuildTargets.cmake:650 (find_arduino_library)
  C:/workspace/arduino/Arduino-CMake-Toolchain-release-1.1-dev/Arduino/System/BoardBuildTargets.cmake:617 (_add_internal_arduino_library)
  C:/workspace/arduino/Arduino-CMake-Toolchain-release-1.1-dev/Arduino/System/BoardBuildTargets.cmake:183 (_link_ard_lib_list)
  CMakeLists.txt:9 (target_link_arduino_libraries)

-- Configuring incomplete, errors occurred!
See also "C:/workspace/ARC003-2/cmake-build-debug/CMakeFiles/CMakeOutput.log".
See also "C:/workspace/ARC003-2/cmake-build-debug/CMakeFiles/CMakeError.log".

[Failed to reload]
```

library.properties:
```
name=ESP32 BLE Arduino
version=1.0.1
author=Neil Kolban <kolban1@kolban.com>
maintainer=Dariusz Krempa <esp32@esp32.eu.org>
sentence=BLE functions for ESP32
paragraph=This library provides an implementation Bluetooth Low Energy support for the ESP32 using the Arduino platform. 
category=Communication
url=https://github.com/nkolban/ESP32_BLE_Arduino
architectures=esp32
includes=BLEDevice.h, BLEUtils.h, BLEScan.h, BLEAdvertisedDevice.h
```

I am going to keep at it, maybe I am missing something but your help would be great.

Thanks

a9183756-gh, 2020-04-25:
I have marked this issue as enhancement to make the upload, ISP and other tool related options (like SERIAL_PORT) available through board options file, to improve the usage experience.

raphael-bmec-co, 2020-04-25:
Thank you, that would be a nice enhancement.

I have another question. Again, not sure if I am doing something wrong.

I am building a large project and I have a .h and .cpp file in some nested folder:
`modules/ble/services/wifi/ble_wifi_services.h` 
`modules/ble/services/wifi/ble_wifi_services.cpp`

CMake runs fine but when compiling I get:

```
CMakeFiles/arc003-2.dir/arc003-2.cpp.o:(.literal._Z8ble_initv+0x20): undefined reference to `BleServiceWifiConfig::BleServiceWifiConfig(BLEServer*)'
CMakeFiles/arc003-2.dir/arc003-2.cpp.o:(.literal._Z8ble_initv+0x24): undefined reference to `BleServiceWifiScanner::BleServiceWifiScanner(BLEServer*)'
CMakeFiles/arc003-2.dir/arc003-2.cpp.o: In function `ble_init()':
C:\workspace\ARC003-2/arc003-2.cpp:512: undefined reference to `BleServiceWifiConfig::BleServiceWifiConfig(BLEServer*)'
C:\workspace\ARC003-2/arc003-2.cpp:513: undefined reference to `BleServiceWifiScanner::BleServiceWifiScanner(BLEServer*)'
collect2.exe: error: ld returned 1 exit status
CMake Error at .scripts/LinkScript.cmake:286 (message):
  Linking arc003-2 failed!!!

mingw32-make.exe[7]: *** [arc003-2.elf] Error 1
mingw32-make.exe[6]: *** [CMakeFiles/arc003-2.dir/all] Error 2
mingw32-make.exe[5]: *** [CMakeFiles/arc003-2.dir/rule] Error 2
CMakeFiles\arc003-2.dir\build.make:89: recipe for target 'arc003-2.elf' failed
CMakeFiles\Makefile2:116: recipe for target 'CMakeFiles/arc003-2.dir/all' failed
CMakeFiles\Makefile2:123: recipe for target 'CMakeFiles/arc003-2.dir/rule' failed
Makefile:130: recipe for target 'arc003-2' failed
mingw32-make.exe[4]: *** [arc003-2] Error 2
CMake Error at .scripts/upload.cmake:153 (message):
  *** upload for 'arc003-2' failed!
Call Stack (most recent call first):
  .scripts/upload.cmake:325 (_build_and_resolve_target)
```

Questions:
1. In my main app, should I include ble_wifi_services.h as:
`#include "modules/ble/services/wifi/ble_wifi_services.h"`
OR
`#include "ble_wifi_services.h"` (when I do it this way the IDE does not find the h file but I'm concerned this is becuase it isn't linking).
2.  Is the CMakeLists.txt below correct?

```
cmake_minimum_required(VERSION 3.0.0)

project(arc003-2 CXX)

#=================================================================
# app
add_executable(arc003-2 arc003-2.cpp)

target_link_arduino_libraries(arc003-2 AUTO_PRIVATE)

# This is needed for the generation of HEX binary and uploading it
target_enable_arduino_upload(arc003-2)

#=================================================================
# drivers

#=================================================================
# modules

# modules/ble/services/wifi
add_executable(ble_wifi_service modules/ble/services/wifi/ble_wifi_services.h modules/ble/services/wifi/ble_wifi_services.cpp)
target_link_arduino_libraries(ble_wifi_service AUTO_INTERFACE modules/ble/services/wifi/ble_wifi_services.h AUTO_PRIVATE)
```