a9183756-gh/Arduino-CMake-Toolchain #20: Feature request: Sketch exceeds program storage space warning.

raphael-bmec-co, 2020-04-27:
Arduino IDE provides the following error and blocks upload. This is very useful and a challenging issue to diagnose when upload takes place anyway.

```
Sketch uses 1311950 bytes (100%) of program storage space. Maximum is 1310720 bytes.
Global variables use 57732 bytes (17%) of dynamic memory, leaving 269948 bytes for local variables. Maximum is 327680 bytes.
Sketch too big; see http://www.arduino.cc/en/Guide/Troubleshooting#size for tips on reducing it.
text section exceeds available space in board
Error compiling for board ESP32 Dev Module.
```

a9183756-gh, 2020-04-28:
Can you please elaborate what the feature request is? Do you need the upload to be blocked if the sketch uses more than 100%, or do you want the upload to happen? Please note that the HEX binary conversion will fail if the the sketch uses more than 100% and hence upload cannot happen. Also the toolchain already prints the size of the compiled executable,  and it can be easily seen that the sketch uses more than 100%.

raphael-bmec-co, 2020-04-28:
Thank you. I actually missed that it printed the size. But the upload does proceed as shown below. And then the micro just perpetually restarts. So "Please note that the HEX binary conversion will fail if the the sketch uses more than 100% and hence upload cannot happen" is not happening. The feature request would be to block the upload if the sketch uses more than 100%.

```
Generating upload image for 'arc003-2'
esptool.py v2.6
Calculating 'arc003-2' size
################## Size Summary ##################
Program Size: 1331058 of 1310720 bytes (101%)
Data Size: 57692 of 327680 bytes (17%)
##################################################
[100%] Built target arc003-2
esptool.py v2.6
Serial port COM9
Connecting....
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: 24:6f:28:79:be:b4
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 8192 bytes to 47...
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 5461.3 kbit/s)...
Hash of data verified.
Compressed 17392 bytes to 11186...
Wrote 17392 bytes (11186 compressed) at 0x00001000 in 0.1 seconds (effective 993.8 kbit/s)...
Hash of data verified.
Compressed 1331168 bytes to 785775...
Wrote 1331168 bytes (785775 compressed) at 0x00010000 in 11.1 seconds (effective 955.4 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 128...
Wrote 3072 bytes (128 compressed) at 0x00008000 in 0.0 seconds (effective 3072.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Built target upload

Build finished
```

arduino-cmake-toolchain, 2020-04-28:
I expected some HEX conversion failure if it exceeded 100%. Thanks for pointing this out. It indeed makes sense to explicitly fail the upload in that case. Should be easier to implement.

a9183756-gh, 2020-05-01:
Fixed in release-1.1-dev branch

closed 2020-05-01 by a9183756-gh