a9183756-gh/Arduino-CMake-Toolchain #24: Unknown CMake command "target_link_arduino_libraries"

hacki11, 2020-05-17:
I'm trying to compile the example project hello-world, but i'm stuck with the issue it can not find
`target_link_arduino_libraries`.

do you have any hints to solve this issue?

```
/home/hacki/.local/share/JetBrains/Toolbox/apps/CLion/ch-0/201.7223.86/bin/cmake/linux/bin/cmake -DCMAKE_BUILD_TYPE=Debug -D ARDUINO_TOOLCHAIN_FILE=~/git/Arduino-CMake-Toolchain/Arduino-toolchain.cmake -G "CodeBlocks - Unix Makefiles" /home/hacki/git/Arduino-CMake-Toolchain/Examples/01_hello_world
-- The CXX compiler identification is GNU 9.3.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at CMakeLists.txt:7 (target_link_arduino_libraries):
  Unknown CMake command "target_link_arduino_libraries".

-- Configuring incomplete, errors occurred!
See also "/home/hacki/git/Arduino-CMake-Toolchain/Examples/01_hello_world/cmake-build-debug/CMakeFiles/CMakeOutput.log".

[Finished]
``

a9183756-gh, 2020-05-18:
This error is due to the toolchain not being used by CMake. Please check the path that you pass to CMake (~/git/Arduino-CMake-Toolchain/Arduino-toolchain.cmake) and whether the content of the file is not empty.

hacki11, 2020-05-18:
Ah!!! thanks!

closed 2020-05-18 by hacki11