a9183756-gh/Arduino-CMake-Toolchain #29: Board options not found in Ubuntu

serjche, 2020-06-07:
Hi,

I've been trying to get the toolchain running on Ubuntu 18.04, but after following the steps on https://github.com/a9183756-gh/Arduino-CMake-Toolchain/blob/master/Examples/README.md
the BoardOptions.cmake file it creates is empty:

```cmake
# Copyright (c) 2020 Arduino CMake Toolchain

###############################################################################
# This is an automatically generated template file for board options.
# You may edit it to comment/uncomment selected board and board options.
# However do not change the structure of this template, which is fixed and 
# any change to the structure gets overwritten.
```

And I suspect it simply is not finding the installed Arduino framework, which I'm not really quite sure is installed properly in the first place. I followed the normal instructions to install it in the main Arduino page. I'm a bit of a newbie on Ubuntu so I'm not quite sure if I have everything configured properly. The installation path of arduino is on /usr/local/bin/arduino.

I have tried the same examples on Windows and it works like a charm, it finds the Arduino installation and boards just fine, and the BoardOptions.cmake is populated properly.

Maybe I'm missing some environment variable, or the path of the Arduino installation is wrong, I'm not sure, or maybe is something completely different.

I'll look into the code to see how it determines the Arduino installation path, but I figured I asked for help anyways.

Thanks!

a9183756-gh, 2020-06-07:
You are probably facing a known issue in Ubuntu as explained below [taken from here](https://ubuntu.com/tutorials/install-the-arduino-ide#2-installing-via-a-tarball).

***Installing via apt**
While there is a package for the Arduino IDE on current APT repositories, it has not been updated for a while. As such, while it is still possible to install the IDE by running sudo apt install arduino, it is not recommended to do so, as asking for support when using outdated software is more difficult.*

Although older arduino installation path is detected by the toolchain, it is not able to enumerate the boards due to incompatibility and thus the empty BoardOptions.cmake.

If this is the case, please remove the older version of arduino (`sudo apt-get autoremove arduino`) and then follow the [tutorial for installing via tarball](https://ubuntu.com/tutorials/install-the-arduino-ide#2-installing-via-a-tarball).

If you extract the tar in non-standard location, you may get the error that *Arduino installation is not found*. In that case, please provide the installation directory using `-D ARDUINO_INSTALL_PATH=/path/to/arduino-<version>`. 

**Note**: Just pushed [PRs](https://github.com/a9183756-gh/Arduino-CMake-Toolchain/pulls) to report warning in case if the installed arduino version is < 1.5, and also to identify the installation paths better in Linux. Although not required, you may use that patch if needed.

serjche, 2020-06-13:
Hi, thank you for your quick response, and sorry for my late response hehe...

I've tested your latest commit on the fix/install_path_dev branch, and it detected the arduino installation paths perfectly.

I forgot to mention I did uninstall and install again as you indicated in your previous comment but it didn't make a difference, but the latest changes did fixed the issue.

It created the BoardOptions.cmake just fine:
[image]

And I was able to generate the build files and compile as well.

Thank you so much!

serjche, 2020-06-13:
Also, I see you are waiting for a review to merge the PR, would you like me to add that? I'm not sure if you have a review procedure, or just approving with comments would be enough?

closed 2020-06-13 by serjche

a9183756-gh, 2020-06-15:
Surprising, because the change was not intended to fix the issue you reported. It worked for you probably because you have another folder named 'arduino*' somewhere, which the toolchain picked up earlier.  Now with the change #30, the toolchain possibly picked up the more relevant folder that worked.
For review, comments should be sufficient. Anyway, will merge the issue as it worked for you.