a9183756-gh/Arduino-CMake-Toolchain #34: The CXX compiler identification is unknown (Optiboot)

hacki11, 2020-06-21:
Hi,
having my Board as mega 2560 everything is fine. All compilers are found and perfectly used.
But now i have burned Optiboot onto my mega and selected the optiboot2560 board.

Now it searches avr-gcc and avr-g++ in paths where not binary is located. avr-gcc and avrg++ are available via Path. I have added CMAKE_CXX_COMPILER and CMAKE_C_COMPILER but no success.

Can you give me any pointers where i could possibly look?
I can't tell if the optiboot files are wrong or there is something with arduino toolchain.

```
hacki@lx-hacki:~/git/homeduino$ cmake -DCMAKE_TOOLCHAIN_FILE=~/git/Arduino-CMake-Toolchain/Arduino-toolchain.cmake -G "CodeBlocks - Unix Makefiles" .
/usr/share/arduino
-- Found Arduino package /home/hacki/.arduino15/package_index.json
-- Found Arduino package /home/hacki/.arduino15/package_optiboot_optiboot-additional_index.json
-- Found Arduino package /usr/share/arduino/hardware/package_index_bundled.json
-- Found Arduino Platform: /usr/share/arduino/hardware/arduino/avr
-- Found Arduino Platform: /home/hacki/.arduino15/packages/Optiboot/hardware/avr/0.8.0
-- Selected Arduino Board: Optiboot on Mega2560 [avr.optiboot2560]
-- The CXX compiler identification is unknown
-- Indexing Arduino libraries for the project
CMake Error at CMakeLists.txt:3 (project):
  The CMAKE_CXX_COMPILER:

    /home/hacki/git/homeduino/_pkg_mgr/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++

  is not a full path to an existing compiler tool.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

-- Configuring incomplete, errors occurred!
See also "/home/hacki/git/homeduino/CMakeFiles/CMakeOutput.log".
See also "/home/hacki/git/homeduino/CMakeFiles/CMakeError.log".
```
Board from Optiboot `set(ARDUINO_BOARD "avr.optiboot2560") # Optiboot on Mega2560`

CMakeError.log
```
Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: /home/hacki/git/homeduino/cmake-build-debug/_pkg_mgr/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ 
Build flags: 
Id flags:  

The output was:
No such file or directory
```

a9183756-gh, 2020-06-21:
I tried compiling for the Optiboot on Mega2560 here, and it seems to be working well.

From the above error message, it looks like the toolchain expects the compiler inside `/usr/share/arduino/hardware/tools/avr` (Location of Arduino installation), but could not find. So it tries to find alternative package manager path in `/home/hacki/git/homeduino/_pkg_mgr/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7` and could not find there as well (Since package manager is not enabled). Toolchain never looks for the compiler from *PATH*  (It is not as per the specification). Where is it located in your installation, that may provide a clue?

You may try providing the *CMAKE_C_COMPILER* and *CMAKE_CXX_COMPILER* paths in the the generated *ArduinoSystem.cmake* in the build directory. However this file can get overwritten whenever *BoardOptions.cmake* is changed and is not a reliable solution.

Alternatively, you may try using package manager, as given below, which downloads the compiler and uses, if not found in the expected locations.
```
cmake -D CMAKE_TOOLCHAIN_FILE=/path/to/Arduino-toolchain.cmake -D ARDUINO_BOARD_MANAGER_URL=https://github.com/Optiboot/optiboot/releases/download/v8.0/package_optiboot_optiboot-additional_index.json /path/to/Examples/01_hello_world
```

hacki11, 2020-06-22:
```
hacki@lx-hacki:~/git/homeduino$ locate avr-g++
/usr/bin/avr-g++
/usr/share/arduino/hardware/tools/avr/bin/avr-g++

acki@lx-hacki:~/git/homeduino$ locate avr-gcc
/usr/bin/avr-gcc
/usr/share/arduino/hardware/tools/avr/bin/avr-gcc
```
If i change avr-gcc and avr-g++ executables in ArduinoSystem.cmake and add the path /usr/share/arduino/hardware/tools/avr/bin to `ARDUINO_FIND_ROOT_PATH` it will work.

hacki11, 2020-06-22:
- removed anything of optiboot from ~/.arduino15/
- updated "Arduino AVR Boards" in Boardmanager in Arduino IDE
- added optiboot package json Arduino additional board manager urls
- installed Optiboot 8.0 in Boardmanager in Arduino IDE

now it will find the right compiler!

hacki11, 2020-06-22:
if i understood upload-network cmake target correctly, if have to create an additional make call with the required environment variables to the upload-network target, right?

like:
[image]

a9183756-gh, 2020-06-23:
Good to know that it worked after you refreshed your Arduino Installation. I still wonder why the alternative package manager did not work for you. In case you need this, you may try with a new build directory, and post the full error logs for analysis.

Network upload depends on the way you achieved network connectivity to the board. Some possible options that may work are
1. [esp-link](https://github.com/jeelabs/esp-link) in which case you may want to look at the [Arduino specific comments](https://github.com/jeelabs/esp-link/issues/394) there. *upload-network* arguments take these parameters. Be aware of the need of 5V/3.3V voltage conversion between Mega and ESP. 
2. [ArduinoOTA](https://github.com/jandrassy/ArduinoOTA) if you are using some Ethernet/WiFi shield. If you copy the platform.local.txt as instructed, and then invoke the clean CMake build. *upload-network* parameters should change appropriately (taking only IP address, port and password).

hacki11, 2020-06-23:
I took option 2.
Had to copy my `platform.local.txt` to `/home/hacki/.arduino15/packages/Optiboot/hardware/avr/0.8.0`.
Then i can use the following make file run configuration with just NETWORK_IP, NETWORK_PASSWORD and TARGET like you said:
[image]

Works:
```
Connecting to board ...  done
Uploading sketch ...  done
Flashing sketch ...  done

Sketch uploaded successfully
Built target upload-network
```

Thanks again, great work!