a9183756-gh/Arduino-CMake-Toolchain #42: Linking error when compiling stm32duino project

hackbeere, 2020-11-13:
When compiling a stm32duino project the linker complains about missing references. I tried all branches. The output is:

/home/user/local/arduino/arduino15/packages/STM32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: lib_arduino_lib_core.a(startup_stm32yyxx.S.obj): in function `LoopFillZerobss':
(.text.Reset_Handler+0x2a): undefined reference to `SystemInit'
/home/user/local/arduino/arduino15/packages/STM32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: lib_arduino_lib_STM32duino_Low_Power.a(STM32LowPower.cpp.o): in function `STM32LowPower::begin()':
STM32LowPower.cpp:(.text._ZN13STM32LowPower5beginEv+0x4): undefined reference to `LowPower_init'
/home/user/local/arduino/arduino15/packages/STM32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: lib_arduino_lib_core.a(HardwareSerial.cpp.o): in function `HardwareSerial::write(unsigned char)':
HardwareSerial.cpp:(.text._ZN14HardwareSerial5writeEh+0x2a): undefined reference to `uart_enable_tx'
/home/user/local/arduino/arduino15/packages/STM32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: HardwareSerial.cpp:(.text._ZN14HardwareSerial5writeEh+0x46): undefined reference to `serial_tx_active'
/home/user/local/arduino/arduino15/packages/STM32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: HardwareSerial.cpp:(.text._ZN14HardwareSerial5writeEh+0x52): undefined reference to `uart_attach_tx_callback'
/home/user/local/arduino/arduino15/packages/STM32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: lib_arduino_lib_core.a(HardwareSerial.cpp.o): in function `HardwareSerial::HardwareSerial(void*, HalfDuplexMode_t)':
HardwareSerial.cpp:(.text._ZN14HardwareSerialC2EPv16HalfDuplexMode_t+0x52): undefined reference to `pinmap_pin'
/home/user/local/arduino/arduino15/packages/STM32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: HardwareSerial.cpp:(.text._ZN14HardwareSerialC2EPv16HalfDuplexMode_t+0x5c): undefined reference to `pinmap_pin'
/home/user/local/arduino/arduino15/packages/STM32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: lib_arduino_lib_core.a(HardwareSerial.cpp.o): in function `HardwareSerial::enableHalfDuplexRx()':
HardwareSerial.cpp:(.text._ZN14HardwareSerial18enableHalfDuplexRxEv+0x28): undefined reference to `uart_enable_rx'
/home/user/local/arduino/arduino15/packages/STM32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: lib_arduino_lib_core.a(board.c.o): in function `init':
board.c:(.text.init+0x2): undefined reference to `hw_config_init'
/home/user/local/arduino/arduino15/packages/STM32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: lib_arduino_lib_core.a(wiring_time.c.o): in function `millis':
wiring_time.c:(.text.millis+0x2): undefined reference to `getCurrentMillis'
collect2: error: ld returned 1 exit status
CMake Error at .scripts/LinkScript.cmake:286 (message):
  Linking test_deepsleep failed!!!

a9183756-gh, 2020-11-14:
I tried the example build with stm32duino, and it seems to be working well. So, I think it might be some simple issue that you are facing. In your case, many of the undefined references point to SrcWrapper library that comes with STM32 platform.

Please ensure that you add target_enable_arduino_upload(test_deepsleep) in your code. Although seems unrelated, it is a workaround for linking some additional files in stm32duino. If you still face issue, please share the verbose build log (make VERBOSE=1).

hackbeere, 2020-11-14:
Thank you very much for the quick help.
The solution was the missing line "target_enable_arduino_upload(test_deepsleep)" in my CMakeLists.txt.
The compilation and linking of all libraries works now perfectly.

closed 2020-11-14 by hackbeere