a9183756-gh/Arduino-CMake-Toolchain #45: Sketch (ino) support broken

galou, 2021-01-25:
Hi,

first of all, thanks for developing this project and trying and avoiding the Arduino "IDE".

I'm trying to compile the Sketch `Blink.ino`:
```
cd /tmp
mkdir act && cd act
cp -r /opt/arduino-1.8.13/examples/01.Basics/Blink/ .
cd Blink
cat <<EOF > CMakeLists.txt
cmake_minimum_required(VERSION 3.10)
project(Blink CXX)
add_executable(blink Blink.ino)
target_link_arduino_libraries(blink PRIVATE core)
EOF
cd ..
mkdir build && cd build
cmake -DCMAKE_TOOLCHAIN_FILE=$HOME/04-sources/arduino-cmake-toolchain/Arduino-toolchain.cmake -DARDUINO_BOARD="Arduino Uno [avr.uno]" ../Blink
make
```

Compilation suceeds but linking fails:
```
/tmp/ccJpGAAX.ltrans0.ltrans.o: In function `main':
/opt/arduino-1.8.13/hardware/arduino/avr/cores/arduino/main.cpp:43: undefined reference to `setup'
/opt/arduino-1.8.13/hardware/arduino/avr/cores/arduino/main.cpp:46: undefined reference to `loop'
collect2: error: ld returned 1 exit status
```

For easier reference, the `CMakeLists.txt` file is
```cmake
cmake_minimum_required(VERSION 3.10)
project(Blink CXX)
add_executable(blink Blink.ino)
target_link_arduino_libraries(blink PRIVATE core)
```
In the case that I'm doing something wrong and that other users might do the same, it would be good to document how to compile sketches.

The workaround to rename `Blink.ino` to `Blink.cpp` may work here, I didn't try it, but I have another project, where Arduino-Cmake-Toolchain doesn't find the Arduino library when using the `cpp` extension.

Arduino-Cmake-Toolchain e745a9bed3c3fb83 (2020-06-16), Arduino 1.8.13.

raphael-bmec-co, 2021-01-25:
Have you included the following in your file with `setup` and `loop`?

```cpp
#include <Arduino.h>
```

galou, 2021-01-25:
I hadn't but adding the line doesn't help.

a9183756-gh, 2021-01-25:
Support for *.ino* files is **not** there at all yet in the toolchain. A simple workaround may be to rename *.ino* to *.cpp* file and including Arduino.h (`#include "Arduino.h"`)

Please report the issue *"Arduino-Cmake-Toolchain not finding the Arduino library when using the cpp extension"*, if that is what is blocking you from renaming *.ino* to *.cpp*. May be you can try the branch *release-1.1-dev* before reporting the issue (`git checkout release-1.1-dev`).

galou, 2021-01-26:
I'm sorry, I didn't see that the support for `*.ino` files was unticked, I just saw that it was in the list of features.

Anyway, the Blink sketch works perfectly when renaming `Blink.ino` to `Blink.cpp`.

I wanted to use Arduino-Cmake-Toolchain for my project, though but didn't succeed. The first step I did is to install support for the OpenCR board in Arduino (adding `https://raw.githubusercontent.com/ROBOTIS-GIT/OpenCR/master/arduino/opencr_release/package_opencr_index.json` to the list of URLs for the Board Manager, according to [the doc](https://emanual.robotis.com/docs/en/parts/controller/opencr10/#install-on-linux) and to install the Dynamixel2Arduino library. Then I did
```bash
#!/bin/bash
mkdir /tmp/turtlebot3_ros2 && cd /tmp/turtlebot3_ros2
cp -r ~/.arduino15/packages/OpenCR/hardware/OpenCR/1.4.17/libraries/turtlebot3_ros2/examples/turtlebot3_burger .
echo << EOF > turtlebot3_burger/CMakeLists.txt
cmake_minimum_required(VERSION 3.10)
project(turtlebot3_burger CXX)
add_executable(turtlebot3_burger turtlebot3_burger.cpp)
target_link_arduino_libraries(turtlebot3_burger PRIVATE core "TurtleBot3 ROS2")
EOF
mv turtlebot3_burger/turtlebot3_burger.ino turtlebot3_burger/turtlebot3_burger.cpp
mkdir build && cd build
cmake -DCMAKE_TOOLCHAIN_FILE=$HOME/04-sources/arduino-cmake-toolchain/Arduino-toolchain.cmake -DARDUINO_BOARD="OpenCR.OpenCR" ../turtlebot3_burger
make
```

With the current HEAD and `-DARDUINO_BOARD="OpenCR Board [OpenCR.OpenCR]"`, cmake fails:
```
CMake Error at /home/gael/04-sources/arduino-cmake-toolchain/Arduino/System/BoardBuildTargets.cmake:472 (message):
  Arduino library TurtleBot3 ROS2 could not be found in

```

and with `release-1.1-dev` and `-DARDUINO_BOARD="OpenCR.OpenCR"`, cmake succeeds but make fails:
```
/tmp/examples/turtlebot3_burger/turtlebot3_burger.cpp:17:29: fatal error: TurtleBot3_ROS2.h: No such file or directory
```

fnevgeny, 2021-03-19:
BTW, I added the following lines to force cmake working with ino files:

```
# Force the .ino file to be treated as C++
set_source_files_properties(${PROJECT_SOURCE_DIR}/my_project.ino
    PROPERTIES LANGUAGE CXX
)
set_source_files_properties(${PROJECT_SOURCE_DIR}/my_project.ino
    PROPERTIES COMPILE_FLAGS "-x c++"
)
```