a9183756-gh/Arduino-CMake-Toolchain #54: Upload fails on avr:leonardo

zeechs, 2022-02-03:
Hello,

First, thank you for sharing your work, it is very useful and appreciated. I used it successfully with an Arduino Uno board.

I am having trouble finding the cause of the upload failure using a Leonardo board. I am following these simple steps to configure, build and upload your hello_world example:
- configure: `cmake .. -D CMAKE_TOOLCHAIN_FILE=path/to/Arduino-toolchain.cmake`
- edit `BoardOptions.cmake` and uncomment the following line:
```cmake
set(ARDUINO_BOARD "Arduino Leonardo [avr.leonardo]") # Arduino Leonardo
```
- configure again: `cmake .`
- build: `make`
- upload:
```
$ make SERIAL_PORT=/dev/ttyACM0 upload-hello_world
[ 89%] Built target _arduino_lib_core
[ 96%] Built target hello_world
[100%] Uploading 'hello_world'

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/home/sylvain/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/etc/avrdude.conf"
         User configuration file is "/home/sylvain/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/ttyACM0
         Using Programmer              : avr109
         Overriding Baud Rate          : 57600
         AVR Part                      : ATmega32U4
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  9000  9000 0x00 0x00
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : butterfly
         Description     : Atmel AppNote AVR109 Boot Loader

Connecting to programmer: .avrdude: butterfly_recv(): programmer is not responding

avrdude: butterfly_recv(): programmer is not responding
avrdude: butterfly_recv(): programmer is not responding
avrdude: butterfly_recv(): programmer is not responding
avrdude: butterfly_recv(): programmer is not responding
avrdude: butterfly_recv(): programmer is not responding
Found programmer: Id = "�"; type =
    Software Version = .; Hardware Version = .
avrdude: butterfly_recv(): programmer is not responding
avrdude: butterfly_recv(): programmer is not responding
avrdude: error: buffered memory access not supported. Maybe it isn't
a butterfly/AVR109 but a AVR910 device?
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: butterfly_recv(): programmer is not responding
avrdude: error: programmer did not respond to command: leave prog mode
avrdude: butterfly_recv(): programmer is not responding
avrdude: error: programmer did not respond to command: exit bootloader
```

I'm not sure where to go from here.
Uploading a sketch with `arduino-cli` works well.

zeechs, 2022-02-07:
This board needs to be switched to bootloader mode by hitting the reset button twice before attempting to upload.
In my case, there is no reset button(DFRobot Beetle). It can be done by opening the serial port at 1200 bauds and then closing it.
The board is then ready for upload.

eigendude, 2022-03-04:
I needed to reset the Leonardo in an automated way, because I'm flashing and testing various Arduinos from a CI pipeline in the cloud.

As @zeechs says, you need to open the serial port (probably `/dev/ttyACM0`) with a baud of 1200, and then close it. The LEDs will flash the same pattern as the reset button while the firmware is flashable. After resetting via the button or open/closing the serial port, the Leonardo is flashable for about 8 seconds.

Here's my GitHub Actions yaml for my flashing and testing:

```yml
jobs:
  build:
    steps:
      - name: Upload Ardino hex to /dev/ttyACM0
        run: |
          ./scripts/reset_leonardo.py /dev/ttyACM0
          sleep 2
          cmake --build ./build --target upload -- TARGET=leonardo_node SERIAL_PORT=/dev/ttyACM0
          sleep 2

      - name: Test Ardino
        run: |
          ./firmata_test.py
```

And here's the contents of `reset_leonardo.py`:

```python
#!/usr/bin/env python3
################################################################################
#
#  Copyright (C) 2021 Garrett Brown
#  SPDX-License-Identifier: Apache-2.0
#
################################################################################

import sys
import serial

# Environment parameters
SERIAL_PORT = sys.argv[1]

# Open the Leonardo tty at 1200 baud and then close to reset into programming
# mode
print(f"Opening {SERIAL_PORT} at 1200 baud")
ser = serial.Serial(SERIAL_PORT, 1200)

print(f"Closing {SERIAL_PORT}")
ser.close()

print("Arduino Leonardo is now in reset mode for 8 seconds")
```