a9183756-gh/Arduino-CMake-Toolchain #62: undefined reference to `setup()'

KSDaemon, 2023-02-24:
Hi all!

I'm trying to setup toolchain. After a bunch of experiments I managed to compile the code, but linking phase is errored.

I'm developing a library and have an examples folder with some `ino` sketches. 

But I have no luck of compiling them. I'm not even talking about developing a library (that's beyond fantasy).

Here is what I get:
```shell
[100%] Linking CXX executable Callee-ESP32.elf
lib_arduino_lib_core.a(main.cpp.o):(.literal._Z8loopTaskPv+0x8): undefined reference to `setup()'
lib_arduino_lib_core.a(main.cpp.o):(.literal._Z8loopTaskPv+0xc): undefined reference to `loop()'
lib_arduino_lib_core.a(main.cpp.o): In function `loopTask(void*)':
/Users/kostik/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/cores/esp32/main.cpp:18: undefined reference to `setup()'
/Users/kostik/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/cores/esp32/main.cpp:21: undefined reference to `loop()'
collect2: error: ld returned 1 exit status
gmake[3]: *** [CMakeFiles/Callee-ESP32.dir/build.make:87: Callee-ESP32.elf] Error 1
gmake[2]: *** [CMakeFiles/Makefile2:593: CMakeFiles/Callee-ESP32.dir/all] Error 2
gmake[1]: *** [CMakeFiles/Makefile2:600: CMakeFiles/Callee-ESP32.dir/rule] Error 2
gmake: *** [Makefile:332: Callee-ESP32] Error 2
```

Here is related part of my CMakeLists.txt:
```
cmake_minimum_required(VERSION 3.8)
project(WampCaster CXX)
set(CMAKE_CXX_STANDARD 11)

# Adding all examples as executables for syntax highlighting and building
add_executable(Callee-ESP32 examples/Callee-ESP32/Callee-ESP32.ino)
set_target_properties(Callee-ESP32 PROPERTIES LINKER_LANGUAGE CXX)
target_link_arduino_libraries(Callee-ESP32 AUTO_PRIVATE)
```

In the board I enabled one of the ESP32 (I tried some other variants, same results):
```
set(ARDUINO_BOARD "DOIT ESP32 DEVKIT V1 [esp32.esp32doit-devkit-v1]") # DOIT ESP32 DEVKIT V1
```

Of course if I try to compile the same sketches with arduino-cli - everything works as expected:
```sh
> /usr/local/bin/arduino-cli compile -b esp32:esp32:esp32doit-devkit-v1 examples/Callee-ESP32/Callee-ESP32.ino
...
Used library       Version Path                                                                                          
WiFi               1.0     /Users/kostik/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/WiFi            
WampCaster         0.1.0   /Users/kostik/Projects/WampCaster                                                             
ArduinoJson        6.20.1  /Users/kostik/Projects/Arduino/libraries/ArduinoJson                                          
WebSockets_Generic 2.16.1  /Users/kostik/Projects/WebSockets_Generic                                                     
WiFiClientSecure   1.0     /Users/kostik/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/WiFiClientSecure
Crypto             0.4.0   /Users/kostik/Projects/Arduino/libraries/Crypto                                               
base64             1.3.0   /Users/kostik/Projects/Arduino/libraries/base64                                               

Used platform Version Path                                                               
esp32:esp32   1.0.6   /Users/kostik/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6
```

If someone knows what it can be or where to look - would love to hear! Thnx!

technyon, 2023-02-24:
Did you try the hello world example, that's a good start.

My guess is you need this:

target_link_arduino_libraries(WampCaster PRIVATE core)

If that doesn't work, can you attach everything in a zip file?

KSDaemon, 2023-02-24:
Hi @technyon! Thnx for a super quick answer! :) 

Well, I tried `hello_world` example as it is - it works. That is great! 
Okay, I figured out that specifying `ino` file in `add_executable` even with `set_target_properties(hello_world PROPERTIES LINKER_LANGUAGE CXX)` doesn't work:

```
add_executable(hello_world examples/01_hello_world/hello_world.ino)
set_target_properties(hello_world PROPERTIES LINKER_LANGUAGE CXX)
target_link_arduino_libraries(hello_world PRIVATE core)
```

Okay. That is not a big problem. I can link `cpp` → to `ino` so it will be possible to compile from CLion and typical Arduino user will be able to run the sketch ino.

But here is the next problem: I just added `#include <WiFiMulti.h>` to `hello_world` example - and it can't find the library. 
But `WiFiMulti` is a core library and is it a part of esp32 core:

```
❯ find ~/Library/Arduino15/packages -name "WiFiMulti.h"
/Users/kostik/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/WiFi/src/WiFiMulti.h
```

What is interesting: when I ran first cmake initialization: it finds all libraries:

```
"/Users/kostik/Library/Application Support/JetBrains/Toolbox/apps/CLion/ch-0/223.8214.51/CLion.app/Contents/bin/cmake/mac/bin/cmake" -D CMAKE_TOOLCHAIN_FILE=./Arduino-CMake-Toolchain/Arduino-toolchain.cmake -S /Users/kostik/Projects/WampCaster -B /Users/kostik/Projects/WampCaster
/Applications/Arduino.app/Contents/Java
-- Found Arduino Platform: /Applications/Arduino.app/Contents/Java/hardware/arduino/avr
-- Found Arduino Platform: /Users/kostik/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6
-- Found Arduino Platform: /Users/kostik/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.1
-- Found Arduino Platform: /Users/kostik/Library/Arduino15/packages/arduino/hardware/avr/1.8.6
-- Found Arduino Platform: /Users/kostik/Library/Arduino15/packages/arduino/hardware/samd/1.8.13
-- Selected Arduino Board: DOIT ESP32 DEVKIT V1 [esp32.esp32doit-devkit-v1]
-- Selected board option: "Flash Frequency" = "80MHz"
-- Selected board option: "Upload Speed" = "921600"
-- Selected board option: "Core Debug Level" = "None"
-- The CXX compiler identification is GNU 5.2.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Users/kostik/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Arduino Library ArduinoJson: /Users/kostik/Projects/Arduino/libraries/ArduinoJson
-- The C compiler identification is GNU 5.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Users/kostik/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Arduino Library base64: /Users/kostik/Projects/Arduino/libraries/base64
-- Found Arduino Library Crypto: /Users/kostik/Projects/Arduino/libraries/Crypto
-- Found Arduino Library WebSockets_Generic: /Users/kostik/Projects/Arduino/libraries/WebSockets_Generic
-- Found Arduino Library WiFi: /Users/kostik/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/WiFi
-- Found Arduino Library WiFiClientSecure: /Users/kostik/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/WiFiClientSecure
-- Found Arduino Library SPI: /Users/kostik/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/SPI
-- Found Arduino Library Ethernet_Generic: /Users/kostik/Projects/Arduino/libraries/Ethernet_Generic
-- Found Arduino Library UIPEthernet: /Users/kostik/Projects/Arduino/libraries/UIPEthernet
-- Found Arduino Library WiFiNINA_Generic: /Users/kostik/Projects/Arduino/libraries/WiFiNINA_Generic
-- Found Arduino Library WiFi101: /Users/kostik/Projects/Arduino/libraries/WiFi101_Generic
-- Found Arduino Library EthernetENC: /Users/kostik/Projects/Arduino/libraries/EthernetENC
-- Found Arduino Library Ethernet: /Users/kostik/Projects/Arduino/libraries/Ethernet
-- Found Arduino Library WiFi101_Generic: /Users/kostik/Projects/Arduino/libraries/WiFi101_Generic
-- Found Arduino Library WampCaster: /Users/kostik/Projects/Arduino/libraries/WampCaster
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/kostik/Projects/WampCaster

[Finished]
```

But when then I try to compile executable - it can't :(
[... 7 lines cut ...]
gmake[3]: *** [CMakeFiles/hello_world.dir/build.make:76: CMakeFiles/hello_world.dir/examples/01_hello_world/hello_world.cpp.o] Error 1
gmake[2]: *** [CMakeFiles/Makefile2:591: CMakeFiles/hello_world.dir/all] Error 2
gmake[1]: *** [CMakeFiles/Makefile2:598: CMakeFiles/hello_world.dir/rule] Error 2
gmake: *** [Makefile:332: hello_world] Error 2
```

Addition of `target_link_arduino_libraries(hello_world AUTO_PUBLIC)` doesn't help :(

btw, @technyon can you explain the difference between `PRIVATE | PUBLIC | AUTO_PRIVATE | AUTO_PUBLIC` keywords. 
I found this description: 

```
	set(keywords
		PRIVATE        # Private-linked Arduino Libraries
		PUBLIC         # Public-linked Arduino Libraries
		INTERFACE      # Interface-linked Arduino Libraries
		AUTO           # Default-linked auto detected Arduino libraries
		AUTO_PRIVATE   # Private-linked auto detected Arduino libraries
		AUTO_PUBLIC    # Public-linked auto detected Arduino libraries
		AUTO_INTERFACE # Interface-linked auto detected Arduino libraries
		IGNORE         # Auto detected library names to be ignored
		OVERRIDE       # Customized targets that overrides auto detection
	)
```
But it is still not clear for me what is the difference between private and public as they both resolve to the same libs.

KSDaemon, 2023-02-24:
UPDATE: 
> Okay. That is not a big problem. I can link cpp → to ino so it will be possible to compile from CLion and typical Arduino user will be able to run the sketch ino.

Seems that is not possible :( Making so breaks the build from Arduino IDE.

technyon, 2023-02-24:
First of all, I'd suggest to upgrade your core to 2.0.6. The Wifi is quite broke in 1.0.6, and if you run over a long time you get random disconnects that aren't even reported as such. For that you need my fork.

I think it's not all that important to run it as .ino, most people actually use vscode with platformio, and .cpp is no problem there. If you really want an .ino file, maybe just include main.cpp from there?

I'm not really sure about the AUTO_ keywords either. Maybe they refer to things that are included in the core?

Also to save you some time, if you need to link both WiFi and DNSServer, you want to add this:

target_link_arduino_libraries(DNSServer PUBLIC WiFi)

KSDaemon, 2023-02-24:
O thnx! Seems that I was using rather old links for the esp32 board. I updated it to the latest!

Btw, I'm using your fork instead of this one! :)

Well.... After update to 2.0.7 ESP32 Core I got new error:

```sh
❯ cmake -D CMAKE_TOOLCHAIN_FILE=./Arduino-CMake-Toolchain/Arduino-toolchain.cmake ./
/Applications/Arduino.app/Contents/Java
-- Found Arduino Platform: /Applications/Arduino.app/Contents/Java/hardware/arduino/avr
-- Found Arduino Platform: /Users/kostik/Library/Arduino15/packages/esp32/hardware/esp32/2.0.7
-- Found Arduino Platform: /Users/kostik/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.1
-- Found Arduino Platform: /Users/kostik/Library/Arduino15/packages/arduino/hardware/avr/1.8.6
-- Found Arduino Platform: /Users/kostik/Library/Arduino15/packages/arduino/hardware/samd/1.8.13
-- Selected Arduino Board: DOIT ESP32 DEVKIT V1 [esp32.esp32doit-devkit-v1]
-- Selected board option: "Flash Frequency" = "80MHz"
-- Selected board option: "Upload Speed" = "921600"
-- Selected board option: "Core Debug Level" = "None"
-- Selected board option: "Erase All Flash Before Sketch Upload" = "Disabled"
-- The CXX compiler identification is unknown
CMake Error at CMakeLists.txt:2 (project):
  The CMAKE_CXX_COMPILER:

    {tools.{build.tarch}-esp32-elf-gcc.path}/bin/xtensa-esp32-elf-g++

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

-- Configuring incomplete, errors occurred!
See also "/Users/kostik/Projects/WampCaster/CMakeFiles/CMakeOutput.log".
See also "/Users/kostik/Projects/WampCaster/CMakeFiles/CMakeError.log".
❯ cat CMakeFiles/CMakeOutput.log
The target system is: Arduino -  -
The host system is: Darwin - 21.6.0 - x86_64
❯ cat CMakeFiles/CMakeError.log
Checking whether the CXX compiler is IAR using "" did not match "IAR .+ Compiler":
Checking whether the CXX compiler is IAR using "" did not match "IAR .+ Compiler":
```

I tried 2.0.6 version - same error :(

I see mb related issue: https://github.com/espressif/esp-idf/issues/3937

I'm on Mac OS 12 Monterey. Brew installed of course.

technyon, 2023-02-24:
Which arduino ide did you install? I tried upgrading to 2.x, but it doesn't work unfortunately. You have to install the 1.x version labelled as Legacy IDE instead. I should put these things into the readme :).

KSDaemon, 2023-02-24:
hah) Well, I have installed both :) 

* Latest Arduino IDE
* Latest legacy Arduino
* Latest CLion (but that doesn't make sense as it uses cmake under the hood)

Arduino-CMake-Toolchain finds and uses the legacy IDE. See here:
`/Applications/Arduino.app/Contents/Java`
```
❯ cmake -D CMAKE_TOOLCHAIN_FILE=./Arduino-CMake-Toolchain/Arduino-toolchain.cmake ./
/Applications/Arduino.app/Contents/Java
-- Found Arduino Platform: /Applications/Arduino.app/Contents/Java/hardware/arduino/avr
-- Found Arduino Platform: /Users/kostik/Library/Arduino15/packages/esp32/hardware/esp32/2.0.7
-- Found Arduino Platform: /Users/kostik/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.1
-- Found Arduino Platform: /Users/kostik/Library/Arduino15/packages/arduino/hardware/avr/1.8.6
-- Found Arduino Platform: /Users/kostik/Library/Arduino15/packages/arduino/hardware/samd/1.8.13
-- Selected Arduino Board: DOIT ESP32 DEVKIT V1 [esp32.esp32doit-devkit-v1]
-- Selected board option: "Flash Frequency" = "80MHz"
-- Selected board option: "Upload Speed" = "921600"
-- Selected board option: "Core Debug Level" = "None"
-- Selected board option: "Erase All Flash Before Sketch Upload" = "Disabled"
-- The CXX compiler identification is unknown
```

KSDaemon, 2023-02-24:
So to reproduce:

* Install latest legacy Arduino
* Install espressif ESP32 2.0.x boards
* init Arduino-CMake-Toolchain within `Examples/01_hello_world`
* uncomment any ESP32 board in BoardOptions.cmake
* try `cmake -D CMAKE_TOOLCHAIN_FILE=./Arduino-CMake-Toolchain/Arduino-toolchain.cmake ./`

and hit:
```
❯ cmake -D CMAKE_TOOLCHAIN_FILE=./Arduino-CMake-Toolchain/Arduino-toolchain.cmake ./
/Applications/Arduino.app/Contents/Java
-- Found Arduino Platform: /Applications/Arduino.app/Contents/Java/hardware/arduino/avr
-- Found Arduino Platform: /Users/kostik/Library/Arduino15/packages/esp32/hardware/esp32/2.0.7
-- Found Arduino Platform: /Users/kostik/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.1
-- Found Arduino Platform: /Users/kostik/Library/Arduino15/packages/arduino/hardware/avr/1.8.6
-- Found Arduino Platform: /Users/kostik/Library/Arduino15/packages/arduino/hardware/samd/1.8.13
-- Selected Arduino Board: DOIT ESP32 DEVKIT V1 [esp32.esp32doit-devkit-v1]
-- Selected board option: "Flash Frequency" = "80MHz"
-- Selected board option: "Upload Speed" = "921600"
-- Selected board option: "Core Debug Level" = "None"
-- Selected board option: "Erase All Flash Before Sketch Upload" = "Disabled"
-- The CXX compiler identification is unknown
CMake Error at CMakeLists.txt:2 (project):
  The CMAKE_CXX_COMPILER:

    {tools.{build.tarch}-esp32-elf-gcc.path}/bin/xtensa-esp32-elf-g++

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.
```

technyon, 2023-02-24:
There's no reason to edit the BoardOptions.cmake before running cmake, you can pick one later (I use ccmake). This is my command line:

cmake -DCMAKE_TOOLCHAIN_FILE=~/workspace/sdk/Arduino-CMake-Toolchain/Arduino-toolchain.cmake -GNinja -DCMAKE_BUILD_TYPE=Release  ..

Also, should cmake succeed, you have to copy one of the .csv partition table files into the cmake folder. This somehow doesn't happen anymore for 2.x.

KSDaemon, 2023-02-24:
Hm... Am I miss something? 

I'm doing as it is written in the docs. And first run of cmake fails with the message `Please select a valid arduino board and its menu options using one of the below methods.`

That's why I have to edit BoardOptions.cmake and rerun cmake again.

technyon, 2023-02-24:
ok I see. Yes, I use ccmake instead of editing boardoptions. Or to skip that step altogether, put this into your CMakeList.txt:

set(ARDUINO_BOARD "ESP32 Dev Module [esp32.esp32]")

technyon, 2023-02-24:
P.S.: What about taking cmakes advice:

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

Not sure why it can't detect the compiler automatically, maybe it's MacOS problem.

KSDaemon, 2023-02-25:
Well... I tried to setup different env vars. Actually nothing helps :( I think its because `{tools.{build.tarch}-esp32-elf-gcc.path}` internal variables are not resolved at all. 

I hardcoded the compiler path right in the `platform.txt` something like that:

```sh
> vim /Users/kostik/Library/Arduino15/packages/esp32/hardware/esp32/2.0.7/platform.txt

#compiler.path={tools.{build.tarch}-{build.target}-elf-gcc.path}/bin/
compiler.path=/Users/kostik/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/
```

Well, that helps to make it one step further. Now cmake says that it can not compile a simple test program:
```
CMake Error at /usr/local/Cellar/cmake/3.25.2/share/cmake/Modules/CMakeTestCXXCompiler.cmake:63 (message):
  The C++ compiler

    "/Users/kostik/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/xtensa-esp32-elf-g++"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /Users/kostik/Projects/WampCaster/build/CMakeFiles/CMakeScratch/TryCompile-UNhQiP

    Run Build Command(s):/usr/local/bin/gmake -f Makefile cmTC_96f46/fast && /usr/local/bin/gmake  -f CMakeFiles/cmTC_96f46.dir/build.make CMakeFiles/cmTC_96f46.dir/build
    gmake[1]: Entering directory '/Users/kostik/Projects/WampCaster/build/CMakeFiles/CMakeScratch/TryCompile-UNhQiP'
    Building CXX object CMakeFiles/cmTC_96f46.dir/testCXXCompiler.cxx.o
    /Users/kostik/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/xtensa-esp32-elf-g++  {compiler.cpreprocessor.flags.{build.mcu}} -I/Users/kostik/Library/Arduino15/packages/esp32/hardware/esp32/2.0.7/tools/sdk/esp32/dio_qspi/include {compiler.cpp.flags.{build.mcu}} -w -Os -DF_CPU=240000000L -DARDUINO=108016 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"ESP32_DEV\"" "-DARDUINO_VARIANT=\"doitESP32devkitV1\"" -DARDUINO_PARTITION_default  -DESP32 -DCORE_DEBUG_LEVEL=0    {build.extra_flags.{build.mcu}} @/Users/kostik/Projects/WampCaster/build/build_opt.h -DARDUINO_ESP32_ESP32DOIT_DEVKIT_V1_MENU_DEBUGLEVEL_NONE -DARDUINO_ESP32_ESP32DOIT_DEVKIT_V1_MENU_ERASEFLASH_NONE -DARDUINO_ESP32_ESP32DOIT_DEVKIT_V1_MENU_FLASHFREQ_80 -DARDUINO_ESP32_ESP32DOIT_DEVKIT_V1_MENU_UPLOADSPEED_921600   /Users/kostik/Projects/WampCaster/build/CMakeFiles/CMakeScratch/TryCompile-UNhQiP/testCXXCompiler.cxx -o CMakeFiles/cmTC_96f46.dir/testCXXCompiler.cxx.o
    xtensa-esp32-elf-g++: error: {compiler.cpreprocessor.flags.{build.mcu}}: No such file or directory
    xtensa-esp32-elf-g++: error: {compiler.cpp.flags.{build.mcu}}: No such file or directory
    xtensa-esp32-elf-g++: error: {build.extra_flags.{build.mcu}}: No such file or directory
    xtensa-esp32-elf-g++: error: /Users/kostik/Projects/WampCaster/build/build_opt.h: No such file or directory
    gmake[1]: *** [CMakeFiles/cmTC_96f46.dir/build.make:78: CMakeFiles/cmTC_96f46.dir/testCXXCompiler.cxx.o] Error 1
    gmake[1]: Leaving directory '/Users/kostik/Projects/WampCaster/build/CMakeFiles/CMakeScratch/TryCompile-UNhQiP'
    gmake: *** [Makefile:127: cmTC_96f46/fast] Error 2

  CMake will not be able to correctly generate this project.
```

For the sake of interest I tried to compile helo_world for esp8266 - hah. It doesn't work too!

```
CMake Error at /usr/local/Cellar/cmake/3.25.2/share/cmake/Modules/CMakeTestCXXCompiler.cmake:63 (message):
  The C++ compiler

    "/Users/kostik/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec/bin/xtensa-lx106-elf-g++"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /Users/kostik/Projects/h1/build/CMakeFiles/CMakeScratch/TryCompile-gEoEMl

    Run Build Command(s):/usr/local/bin/gmake -f Makefile cmTC_4913c/fast && /usr/local/bin/gmake  -f CMakeFiles/cmTC_4913c.dir/build.make CMakeFiles/cmTC_4913c.dir/build
    gmake[1]: Entering directory '/Users/kostik/Projects/h1/build/CMakeFiles/CMakeScratch/TryCompile-gEoEMl'
    Building CXX object CMakeFiles/cmTC_4913c.dir/testCXXCompiler.cxx.o
    /Users/kostik/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec/bin/xtensa-lx106-elf-g++  -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -D_GNU_SOURCE -DESP8266 @/Users/kostik/Projects/h1/build/core/build.opt -I/Users/kostik/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.1/tools/sdk/include -I/Users/kostik/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.1/tools/sdk/lwip2/include -I/Users/kostik/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.1/tools/sdk/libc/xtensa-lx106-elf/include -I/Users/kostik/Projects/h1/build/core -c @/Users/kostik/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.1/tools/warnings/none-g++  -Os -g -free -fipa-pta -Werror=return-type -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++17 -MMD -ffunction-sections -fdata-sections -fno-exceptions  -DMMU_IRAM_SIZE=0x8000 -DMMU_ICACHE_SIZE=0x8000  -DNONOSDK22x_190703=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0   -DARDUINO=108016 -DARDUINO_ESP8266_NODEMCU_ESP12E -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_NODEMCU_ESP12E\"" "-DARDUINO_BOARD_ID=\"{_id}\"" -DLED_BUILTIN=2 -DFLASHMODE_DIO   -DARDUINO_ESP8266_NODEMCUV2_MENU_BAUD_115200 -DARDUINO_ESP8266_NODEMCUV2_MENU_DBG_DISABLED -DARDUINO_ESP8266_NODEMCUV2_MENU_EESZ_4M2M -DARDUINO_ESP8266_NODEMCUV2_MENU_EXCEPTION_DISABLED -DARDUINO_ES [... cut ...]
Thnx @technyon for trying to help! If you or someone else have any ideas - please suggest them. But as I understand you're not on MacOS and do not face the same problems. So it is hard to research.

In the meantime, I'll try to setup platformio. mb that would work somehow. 

I really do not understand how people are creating libraries and even big project sketches within Arduino IDE (event v2). It's so ugly. Well, after looking at sources of some 3rd party libraries - you're not surprised anymore %)

technyon, 2023-02-25:
Unfortunately I'm not on MacOS. You could use a VM with Linux, that's what I'm doing actually on a Windows system. The Arduino IDE is as basic as it gets, no idea how anyone can work with it. If you suse platformio, you're unfortunately stuck with vscode which isn't much better. There's a plugin for clion for platformio, but it doesn't really work, you want cmake for clion.

If you want to try a VM, you could download the one I've setup to compile NUKI Hub:

https://github.com/technyon/nuki_hub

Scroll to the bottom of the readme there's a download link. Setting up a Ubuntu VM is quite easy though.

KSDaemon, 2023-02-25:
> The Arduino IDE is as basic as it gets, no idea how anyone can work with it. If you use platformio, you're unfortunately stuck with vscode which isn't much better. 

Yeah! I personally use vscode just like a kind of smart text editor for quick editing standalone files. But that's my imho.

> There's a plugin for clion for platformio, but it doesn't really work

:((( That's sad. I was hoping that is works...

> If you want to try a VM, you could download the one I've setup to compile NUKI Hub:

O great! Thank you! I'll have a look!

KSDaemon, 2023-02-27:
I can't calm down :)
@technyon btw, I tried the `release-1.1-dev` branch and it allows me to build for `esp8266` boards without errors :) 
So I will try to get it to work with esp32... Digging into cmake docs, Arduino build-process docs etc.....

KSDaemon, 2023-02-28:
Closing, as with `release-1.1-dev` and some patches from openned PRs made it works.

closed 2023-02-28 by KSDaemon

technyon, 2023-02-28:
Is it committed somewhere?

KSDaemon, 2023-03-01:
> Is it committed somewhere?

yeah. Here: https://github.com/KSDaemon/Arduino-CMake-Toolchain