a9183756-gh/Arduino-CMake-Toolchain #65: Query: do you know if this works with the Arduino setup on Mac OS?

schnoberts1, 2024-02-02:
It didn't work out of the box for me, but I expect this is an error between chair and keyboard. However, if this hasn't been tested on Mac OS then I'll have a look at the cmake code as well.

KSDaemon, 2024-02-02:
Just a quick remark — it works on Mac OS, believe me :)

KSDaemon, 2024-02-02:
@schnoberts1 And please try one of the more updated forks:
* https://github.com/KSDaemon/Arduino-CMake-Toolchain
* https://github.com/technyon/Arduino-CMake-Toolchain

schnoberts1, 2024-02-02:
Thanks for the redirect :)

closed 2024-02-02 by schnoberts1

schnoberts1, 2024-02-03:
... for the record, I found the issue that prevented the examples working. On my selected platform (Giga R1) the Wire library associated with giga-mbed 4.10.0 does not have a library.properties file so the CMake solution skips over them and then the Wire dependent examples fail to build. In fact there's a few "libraries" with this issue despite them being in the GitHub source.