a9183756-gh/Arduino-CMake-Toolchain #66: Constants not defined during compilation

FalcoGer, 2024-02-14:
I am trying to compile for an avr.micro and to use the HID-Project library.

When I try to compile with this CMake file

```cmake
cmake_minimum_required(VERSION 3.0.0)

set(CMAKE_TOOLCHAIN_FILE "/home/user/repositories/arduino/Arduino-CMake-Toolchain/Arduino-toolchain.cmake")

set(ARDUINO_INSTALL_PATH "/home/user/repositories/arduino/arduino-ide/build")
set(CMAKE_EXPORT_COMPILE_COMMANDS ON)

project (AH64MFDContorller CXX)

add_executable(AH64MFDContorller lnk/MFDController.cpp)

target_link_arduino_libraries(AH64MFDContorller PRIVATE HID HID-Project core)

set_target_properties(AH64MFDContorller PROPERTIES LINKER_LANGUAGE CXX)

target_enable_arduino_upload(AH64MFDContorller)
```

I get those errors

```text
In file included from /home/user/Documents/Projects/arduino/AH64MFD/MFDController/lnk/MFDController.cpp:1:0:
/home/user/Documents/Projects/arduino/libraries/HID-Project/src/HID-Project.h:31:2: error: #error HID Project requires Arduino IDE 1.6.7 or greater. Please update your IDE.
 #error HID Project requires Arduino IDE 1.6.7 or greater. Please update your IDE.
  ^~~~~
/home/user/Documents/Projects/arduino/libraries/HID-Project/src/HID-Project.h:35:2: error: #error HID Project can only be used with an USB MCU.
 #error HID Project can only be used with an USB MCU.
  ^~~~~
```

However when I define them, I get warning about redefinitions.

technyon, 2024-02-14:
A few points:

- I take you are using Arduino IDE 1.6.7
- I'm not familiar with the AH64. It has USB HID?
- AH64MFDContorller looks like a typo, i guess it's AH64MFDController

FalcoGer, 2024-02-14:
I'm using Arduino IDE 1.8.20
AH64 is an attack helicopter. I'm trying to make an MFD game controller for it to use in DCS World (A flight sim)
Thanks. I fixed the typo, but that wasn't the actual problem.

I am using this CMake file and it seems to work.

```cmake
cmake_minimum_required(VERSION 3.0.0)

add_compile_definitions(ARDUINO=10820)
add_compile_definitions(USBCON=1)

set(CMAKE_TOOLCHAIN_FILE "/home/paul/repositories/arduino/Arduino-CMake-Toolchain/Arduino-toolchain.cmake")

set(ARDUINO_INSTALL_PATH "/home/paul/repositories/arduino/arduino-ide/build")
set(CMAKE_EXPORT_COMPILE_COMMANDS ON)

project (AH64MFDController CXX)

add_executable(AH64MFDController lnk/MFDController.cpp)

target_link_arduino_libraries(AH64MFDController PRIVATE HID HID-Project core)

set_target_properties(AH64MFDController PROPERTIES LINKER_LANGUAGE CXX)

target_enable_arduino_upload(AH64MFDController)
```

technyon, 2024-02-15:
Sorry I was confused about the AH64 :D.

The question is what's your target board, what hardware are you planning to run this on? I was getting at if it has USB HID functionality, but if settings those constants work, I guess yes.

FalcoGer, 2024-02-15:
I'm using an arduino pro micro. Flashing with the arduino ide works just fine. I selected avr.micro.

I actually just got done debugging and testing the thing on a breadboard and it works fine, even in the flight sim. So it definitely has HID capability.

technyon, 2024-02-17:
Hi,

I could replicate the issue. I'm a bit confused though, i don't have any hid-project library. Did you install any additional packages in the arduino ide?´ that provide this library?

technyon, 2024-02-17:
I get compilation errors like

AH64MFD-Arduino/lnk/MFDController.cpp:164:41: error: 'PIN_A0' was not declared in this scope

Where are those definitions supposed to come from?

technyon, 2024-02-17:
I've added setting the ARDUINO symbol to the toolchain, it's one error less. Not sure how to get the USBCON symbol, it would need to come from the board definition.

FalcoGer, 2024-02-18:
Strange. it compiles fine for me.
I run cmake-gui and I have to configure 3 times before all the errors disappear and then I hit generate. I leave the programmer blank. Though flashing is another issue.

Honestly, I just want to use nvim to write my code. I don't mind compiling and uploading with the arduino thing. But that IDE is a joke.

technyon, 2024-02-19:
The Arduino IDE only helps you installing the SDK files, besides that it's just a text editor. I went for cmake because it's a powerful build system, and it allows me to use CLion instead of the Arduino IDE.

So your issue is fixed now?

FalcoGer, 2024-02-19:
I am using your fork. And no, it seems it defines `ARDUINO=10000´ and it complains about the version being wrong. I still have to re-define it.

technyon, 2024-02-21:
I'm a but confused, did you try with my latest changes? For me it did define a number that was bigger than the required minimum version, so I could compile without redefining it.

FalcoGer, 2024-02-24:
When I look at compile_commands.json, then `-DARDUINO=100000`