a9183756-gh/Arduino-CMake-Toolchain #67: ARDUINO_INSTALL_PATH not found in IDE 2.3.0 on MacOS

myklemykle, 2024-05-24:
Hi, 

When I try to use this tool with IDE 2.3.0, I get errors about ARDUINO_INSTALL_PATH not found. 

It appears that cmake is searching for a file `lib/version.txt`, which exists in my Arduino 1.8 installation (under `Java/`) but does not exist in my Arduino 2.3.0 installation (where there is no `Java/`).  

Is this supposed to work with IDE 2.3.0 & later ?

Thanks,
-mykle-

technyon, 2024-05-25:
You indeed have to use IDE 1.x. Check my somewhat better maintained fork of this project:

https://github.com/technyon/Arduino-CMake-Toolchain

You still have to use IDE 1.x, but in case you want to target ESP32 at least you can use Arduino Core for ESP32 2.x. If you use the unforked project, only 1.x will be possible as well.

myklemykle, 2024-05-25:
OK, good to know. Thanks!

myklemykle, 2024-05-25:
Actually I am still having a problem with the resulting makefiles. Where is the best place for me to ask questions? I was going to open an issue on your fork, now that I'm using that, but I think the issue-tracker is turned off there?

myklemykle, 2024-05-25:
I'm trying to make this work with the RP2040 core from Earl Philhower. I'm just trying to build the examples at the moment.`cmake`  completes, but then during `make` I get this error:
```
ld: cannot open linker script file /Volumes/External/mykle/Documents/tmp/Arduino-CMake-Toolchain/Examples_build/memmap_default.ld: No such file or directory
```
However the build process does create ` /Volumes/External/mykle/Documents/tmp/Arduino-CMake-Toolchain/Examples_build/01_hello_world/memmap_default.ld`. But then later it seems to search for that file in the wrong location, one directory up.

If I link that file from where `make` put it to where `make` is looking for it, then i hit this second error:
```
ld: cannot find /Volumes/External/mykle/Documents/tmp/Arduino-CMake-Toolchain/Examples_build/boot2.o: No such file or directory
```
`boot2.o` is the second-stage flash-RAM configuration routine, very specific to rp2040 I think. But the installed rp2040 core does supply that file. (It also supplies a version of `memmap_default.ld`.)  So maybe there's just a search path missing somewhere?

technyon, 2024-05-26:
Hi,

I've opened the issue tracker on my fork. Those questions are very specific for the rp2040, I haven't really looked into that. Does it work if you manually copy or link boot2.o?

myklemykle, 2024-05-26:
memmap_default.ld and boot2.0 are both created in the example subdirectory 01_hello_world. If I link them both into the main build directory, the next error I get is this:
```
/Volumes/External/mykle/Library/Arduino15/packages/rp2040/tools/pqt-gcc/2.2.0-d04e724/bin/../lib/gcc/arm-none-eabi/12.3.0/../../../../arm-none-eabi/bin/ld: CMakeFiles/hello_world.dir/hello_world.cpp.o: in function `setup':
/Volumes/External/mykle/Documents/tmp/Arduino-CMake-Toolchain/Examples/01_hello_world/hello_world.cpp:7: undefined reference to `_ZN17Adafruit_USBD_CDC5beginEm'
/Volumes/External/mykle/Library/Arduino15/packages/rp2040/tools/pqt-gcc/2.2.0-d04e724/bin/../lib/gcc/arm-none-eabi/12.3.0/../../../../arm-none-eabi/bin/ld: /Volumes/External/mykle/Documents/tmp/Arduino-CMake-Toolchain/Examples/01_hello_world/hello_world.cpp:16: undefined reference to `Serial'
```
This looks like maybe the object file for the Adafruit TinyUSB library is not getting built? That's also a part of the rp2040 core.

myklemykle, 2024-05-28:
Moved discussion to [the fork.](https://github.com/technyon/Arduino-CMake-Toolchain/issues/3)